gschorcht/i2c-ch341-usb #8: No new i2c device detected

antoninhrlt, 2022-03-03:
I've followed the installation instructions.
My device is detect as "usb" when I type: `lsusb` in the command line:
```
Bus 001 Device 008: ID 1a86:5512 QinHeng Electronics CH341 in EPP/MEM/I2C mode, EPP/I2C adapter
```
But when I type: `ls /dev/ | grep "i2c"` there is no new i2c device than before.
I tried to reinstall, reboot... but nothing has changed.
Someone can help me please ?
I don't know if this problem from the driver or my i2c to USB adapter but my connections might good

antoninhrlt, 2022-03-04:
> lsmod | grep ch341 and if present try to rmmod ch341
> 
> +modprobe i2c-dev

Thank you for your reply, but the first command don't print anything even if I do `modprobe i2c-dev` before

xboxplayer9889, 2022-03-04:
if empty answer arrived, you haven't loaded i2c-ch341-usb driver.
it make and install success 
`modprobe i2c-ch341-usb
modprobe i2c-dev`
to start, if not really installed, try insmod i2c-ch341-usb.ko

antoninhrlt, 2022-03-04:
"make" commands worked fine.
I already did `insmod i2c-ch341-usb.ko` and `modprobe i2c-ch341-usb; modprobe i2c-dev` but nothing changed,

`lsmod | grep ch34` still return an empty answer

antoninhrlt, 2022-03-04:
Oh ! I didn't seen that `sudo make` has returned some warnings:
```
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
make -C /lib/modules/5.16.5-200.fc35.x86_64/build M=/home/anton/i2c-ch341-usb  modules
make[1]: Entering directory '/usr/src/kernels/5.16.5-200.fc35.x86_64'
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
  CC [M]  /home/anton/i2c-ch341-usb/i2c-ch341-usb.o
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
  MODPOST /home/anton/i2c-ch341-usb/Module.symvers
  CC [M]  /home/anton/i2c-ch341-usb/i2c-ch341-usb.mod.o
  LD [M]  /home/anton/i2c-ch341-usb/i2c-ch341-usb.ko
  BTF [M] /home/anton/i2c-ch341-usb/i2c-ch341-usb.ko
make[1]: Leaving directory '/usr/src/kernels/5.16.5-200.fc35.x86_64'
```
(but the command didn't failed)

And the second command `sudo make install` returned:
```
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Module i2c-ch341-usb is installed ... uninstall it first
make[1]: Entering directory '/home/anton/i2c-ch341-usb'
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
dkms remove -m i2c-ch341-usb -v 1.0.0 --all
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Module i2c-ch341-usb-1.0.0 for kernel 5.16.5-200.fc35.x86_64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

i2c-ch341-usb.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.16.5-200.fc35.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod.......
Deleting module i2c-ch341-usb-1.0.0 completely from the DKMS tree.
rm -rf /usr/src/i2c-ch341-usb-1.0.0
make[1]: Leaving directory '/home/anton/i2c-ch341-usb'
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
Deprecated feature: MODULES_CONF_ALIAS_TYPE
readelf: /lib/modules/5.16.5-200.fc35.x86_64/build/vmlinux: Error: Not an ELF file - it has the wrong magic bytes at the start
[... cut ...]
Creating symlink /var/lib/dkms/i2c-ch341-usb/1.0.0/source -> /usr/src/i2c-ch341-usb-1.0.0

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.16.5-200.fc35.x86_64 -C /lib/modules/5.16.5-200.fc35.x86_64/build M=/var/lib/dkms/i2c-ch341-usb/1.0.0/build modules...
cleaning build area...

i2c-ch341-usb.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.16.5-200.fc35.x86_64/extra/
depmod......
```

maraya, 2023-08-24:
Same issue here!, but compiles without warnings on Debian x64 Kernel 5.10.0-22  

`[ 4289.703858] i2c-ch341-usb 1-1.2:1.0: ch341_usb_probe: connect device
[ 4289.703867] i2c-ch341-usb 1-1.2:1.0: ch341_usb_probe: bNumEndpoints=3
[ 4289.703872] i2c-ch341-usb 1-1.2:1.0: ch341_usb_probe:   endpoint=0 type=2 dir=1 addr=2
[ 4289.703878] i2c-ch341-usb 1-1.2:1.0: ch341_usb_probe:   endpoint=1 type=2 dir=0 addr=2
[ 4289.703882] i2c-ch341-usb 1-1.2:1.0: ch341_usb_probe:   endpoint=2 type=3 dir=1 addr=1
[ 4289.703890] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: output gpio0 gpio=0 irq=0 
[ 4289.703896] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: output gpio1 gpio=1 irq=1 
[ 4289.703901] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: output gpio2 gpio=2 irq=2 
[ 4289.703906] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: output gpio3 gpio=3 irq=3 
[ 4289.703910] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: input  gpio4 gpio=4 irq=4 (hwirq)
[ 4289.703915] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: input  gpio5 gpio=5 irq=5 
[ 4289.703919] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: input  gpio6 gpio=6 irq=6 
[ 4289.703923] i2c-ch341-usb 1-1.2:1.0: ch341_cfg_probe: input  gpio7 gpio=7 irq=7 
[ 4289.703927] i2c-ch341-usb 1-1.2:1.0: ch341_i2c_probe: start
[ 4289.704040] i2c-ch341-usb 1-1.2:1.0: ch341_i2c_probe: created i2c device /dev/i2c-7
[ 4289.704046] i2c-ch341-usb 1-1.2:1.0: ch341_i2c_set_speed: Change i2c bus speed to 100 kbps
[ 4289.704254] i2c-ch341-usb 1-1.2:1.0: ch341_i2c_probe: done
[ 4289.704259] i2c-ch341-usb 1-1.2:1.0: ch341_irq_probe: start
[ 4289.704363] i2c-ch341-usb 1-1.2:1.0: ch341_irq_probe: irq_base=34
[ 4289.704369] i2c-ch341-usb 1-1.2:1.0: ch341_irq_probe: done
[ 4289.704371] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_probe: start
[ 4289.704386] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=0 dir=0
[ 4289.704388] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=1 dir=0
[ 4289.704390] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=2 dir=0
[ 4289.704392] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=3 dir=0
[ 4289.704394] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=4 dir=1
[ 4289.704396] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=5 dir=1
[ 4289.704398] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=6 dir=1
[ 4289.704401] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=7 dir=1
[ 4289.704699] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_probe: registered GPIOs from 504 to 511
[ 4289.704707] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=0 dir=0
[ 4289.704726] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=0 irq=34
[ 4289.704759] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=1 dir=0
[ 4289.704774] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=1 irq=35
[ 4289.704801] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=2 dir=0
[ 4289.704817] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=2 irq=36
[ 4289.704848] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=3 dir=0
[ 4289.704862] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=3 irq=37
[ 4289.704889] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=4 dir=1
[ 4289.704902] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=4 irq=38
[ 4289.704926] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=5 dir=1
[ 4289.704939] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=5 irq=39
[ 4289.704966] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=6 dir=1
[ 4289.704978] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=6 irq=40
[ 4289.705004] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_get_direction: gpio=7 dir=1
[ 4289.705018] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_to_irq: gpio=7 irq=41
[ 4289.705216] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_probe: done
[ 4289.705247] i2c-ch341-usb 1-1.2:1.0: ch341_usb_probe: connected
[ 4289.705424] i2c-ch341-usb 1-1.2:1.0: ch341_gpio_poll_function: start
`

With DEBUG activated (modified in i2c-ch341-usb.c, line 20) says /dev/i2c-7 is created but it doesn't really exists!